petercorke/robotics-toolbox-matlab #21: A bug to RTB10.1

marooncn, 2017-10-05:
When I use RIB10.1 to plot my robot, there is an error：
> Prismatic joint(s) present: requires the 'workspace' option
I know it 's because there is a prismatic joint in my robot, but when I add workspace option the error still exists.  At last I return back to RTB9.10 and excute the code, there is no bug this time. And I refer to the manual the fuction is the same as the previous version. So I think maybe it's a bug in the new version.
My code lists as below.
` L(1) = Link([0,0.363,0.300,0,0],'standard'); 
L(2) = Link([0,0,0.260,pi,0],'standard'); 
L(3) = Link([0,0,0,0,1],'standard');
L(3).qlim = [0,0.3];
L(4) = Link([0,0,0,0,0],'standard');
RobotArm = SerialLink(L,'name','SCARA') 
RobotArm.plot([0 0 0 0],'workspace',[-1 1 -1 1 -1 1]) `
[image]
The effect of  RTB9.10

Johnmper, 2017-10-15:
Reading the library code i don't really understand the reason to output an error in case an prismatic joint is found, since the variable `reach` is calculated after the error checking for prismatic joints. 
Just commented the code in the library, and it works fine for me.
```
(line 753-755)
%if any(L.isprismatic)
%    error('Prismatic joint(s) present: requires the ''workspace'' option');
%end
```

```
        (...)
            if isempty(opt.reach)
                %
                % simple heuristic to figure the maximum reach of the robot
                %
                
                assert(~isempty(robot), 'RTB:RTBPlot:plot_options', 'robot must be defined to estimate reach');
                L = robot.links;
                %if any(L.isprismatic)
                %    error('Prismatic joint(s) present: requires the ''workspace'' option');
                %end
                reach = 0;
                for i=1:robot.n
                    if L(i).isrevolute
                        reach = reach + abs(L(i).a) + abs(L(i).d);
                    else
                        reach = reach + abs(L(i).a) + max(abs(L(i).qlim));
                    end
                end
                reach = reach + sum(abs(robot.tool.t));
            else
                reach = opt.reach;
            end
        (...)
```

Johnmper, 2017-10-15:
Can also add manually the reach option in plot function, although it isn't in the documentation the function tb_optparse accepts it just fine. (In case you don't want to change library code)

```
L(1) = Link([0,0.363,0.300,0,0],'standard');
L(2) = Link([0,0,0.260,pi,0],'standard');
L(3) = Link([0,0,0,0,1],'standard');
L(3).qlim = [0,0.3];
L(4) = Link([0,0,0,0,0],'standard');
RobotArm = SerialLink(L,'name','SCARA')
RobotArm.plot([0 0 0 0],'workspace',[-1 1 -1 1 -1 1],'reach',1)
```

ErikvOene, 2019-04-24:
I think this issue is fixed.
At least I was not able to reproduce this error (in master branch) using the following 

```
L(1) = Link([0,0.363,0.300,0,0],'standard');
L(2) = Link([0,0,0.260,pi,0],'standard'); 
L(3) = Link([0,0,0,0,1],'standard');
L(3).qlim = [0,0.3];
L(4) = Link([0,0,0,0,0],'standard');
RobotArm = SerialLink(L,'name','SCARA') 
RobotArm.plot([0 0 0 0],'workspace',[-1 1 -1 1 -1 1])
```

closed 2019-11-03 by petercorke