petercorke/robotics-toolbox-matlab #25: Octave issues

doctorcolossus, 2018-03-03:
I'm attempting to install version 10.2 of the Toolbox on the latest Octave 4.2.1 in Arch Linux.

I've never used MatLab, Octave, or the Robotics Toolbox before, so I hope I'm not overlooking anything obvious.

I followed the directions in the readme, extracting rvctools to /usr/share/octave/4.2.1/m.
I then copied '@Link', '@SerialLink', and '@Quaternion' to /usr/share/octave/4.2.1/m/rvctools/robot.

Calling c as suggested in the readme yields the following error:

```
>> mdl_puma560
error: class not found: Link
error: called from
    mdl_puma560 at line 78 column 6
```

I tried renaming '@Link', '@SerialLink', and '@Quaternion' to 'Link', 'SerialLink', and 'Quaternion' respectively. Then the classes seem to be recognized, but this results in a series of warnings on startup:
```
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Link/subsasgn.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Link/subsref.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Link/char.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/double.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/plus.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/minus.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/dot.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/mrdivide.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/norm.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/mtimes.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/inv.m shadows a built-in function
warning: function /usr/share/octave/4.2.1/m/rvctools/robot/Quaternion/mpower.m shadows a built-in function
```
I am not sure if this is the expected behavior or if I have broken something...

However, now the `mdl_puma560` command gets a little further, finding the class, but returning a different error:
```
>> mdl_puma560
parse error near line 261 of file /usr/share/octave/4.2.1/m/rvctools/robot/Link.m

  syntax error

>>>                     p = properties(this(j));
                                     ^

error: called from
    mdl_puma560 at line 78 column 6
```
I am a complete beginner, so I don't know what is wrong with that syntax.
I am not sure if the problem stems only from my ignorance, or perhaps from some change in Octave since the Toolbox patch was initially released, since that was now six years ago.

Thanks in advance for any support.

sashan13, 2018-03-24:
I ran into the same error. Would be nice to at least have some tips on how to fix it.

petercorke, 2018-12-04:
I just don’t have the resources to keep the Octave port alive, but would be happy to accept contributions.  Check out this project 
https://github.com/gedeschaines/robotics-toolbox-python/tree/update1/Octave which is renovating the old python and Octave code.

closed 2018-12-04 by petercorke

IronFist16, 2020-02-26:
Anyone managed to fix this issue?