petercorke/robotics-toolbox-matlab #38: Fix Octave port

petercorke, 2018-12-05:
Get RTB 10.x working with Octave

petercorke, 2018-12-05:
Good work happening @gedeschaines to be integrated into RTB

gedeschaines, 2018-12-14:
I just updated the following files in [./Octave/rvctools](https://github.com/gedeschaines/robotics-toolbox-python/tree/update1/Octave):

common/plot2.m
robot/trplot.m
robot/trplot2.m
robot/utils/hgtransform.m

to address the problem using Octave 4.4.1 hgtransform on 2D and 3D plot lines in the trplot2 and trplot functions. The changes to these files are compatible with RTB v9.8 in MATLAB R2018b. The updates do not address problem with hgtransform on text.

petercorke, 2018-12-16:
Gary,

I’m chatting with somebody about hgtransform issues at 

https://savannah.gnu.org/bugs/?50466 <https://savannah.gnu.org/bugs/?50466>

peter

gedeschaines, 2018-12-17:
Peter,

To put it succinctly, an hgtransform 'parent' property/value pair passed as
arguments in line(), plot() and plot3() functions are interpreted by
Octave's libinterp/corefnc/graphics routines as an hggroup 'parent'
property/value pair; consequently not producing the desired results. I have
sent you an email with a detailed explanation of the bug #50466 issue.

Gary

petercorke, 2018-12-18:
Thanks for posting this on savannah, let’s see what Rik5 says.