petercorke/robotics-toolbox-matlab #39: Python port

petercorke, 2018-12-05:
There are a few disparate projects, eg. @gedeschaines, @adityadua24 to be pulled together and shipped.

adityadua24, 2018-12-05:
@petercorke Sent you an email.

daniel-s-ingram, 2018-12-09:
I would like to help with this. @adityadua24's looks to be the furthest along, so is that where I should focus my efforts?

petercorke, 2018-12-09:
Hi Daniel,

can you reply to me directly by email, would like to share some documentation to your email address. Looks like we have a few like-minded souls interested in this and we should coordinate efforts.

peter

daniel-s-ingram, 2018-12-10:
Sure! What is the best email for you?

gedeschaines, 2018-12-10:
My apologies for not replying sooner, as I was neck deep in software
porting issues due to computer hardware and operating system upgrades. I am
refocusing my attention toward RTB for Python - Update 1 and would like to
coordinate my efforts with you. Specific areas of concern are code
structure, robot pose rendering, automated testing and third-party robotics
APIs.

Although I plan to acquire MATLAB for Linux and RTB for MATLAB shortly, I
may have to continue modifying RTB v9.8 Octave scripts in order to evaluate
RTB for Python modifications. I too understand that Octave 4.4.x has
limited capability to handle MATLAB objects and GUI implementations, and
consequently have not addressed these areas.

Gary

petercorke, 2018-12-11:
Thanks Gary, can you email me at peter.i.corke@gmail.com <mailto:peter.i.corke@gmail.com> from your email account, not via GitHub,  Thx,

petercorke, 2018-12-11:
Thanks Daniel,  can you email me at peter.i.corke@gmail.com <mailto:peter.i.corke@gmail.com> from your email account, not via GitHub,  Thx,

adityadua24, 2018-12-11:
@daniel-s-ingram I will start development on Robopy again in late January. Meanwhile, you can go through the code and if you have any questions regarding existing implementation, just get in touch with me. adityadua24@gmail.com

gedeschaines, 2018-12-17:
I am reviewing the 'Porting Python to RTB' document, but withholding comments while developing a more complete comprehension of the porting project's scope and details with regards to open source solutions and avoiding "reinventing the wheel" syndrome.