petercorke/robotics-toolbox-matlab #42: Cant find DStarPO function ds.path

gabestreit, 2019-02-05:
Hi, I'm trying to use the DStarPO algorithm. Everything works woderfully until I call the "path" method with ds.path(start) as done in the example. Whenever I try this, I get an error saying that the "path" method doesn't exist. I read through the code in both DStarPO and other related files and can't seem to find where "path" is. Would it be possible to give me some direction as to where I can find "path", and how I can implement it?

petercorke, 2019-02-05:
This one was contributed to RTB9  but I lost touch with the author.  I haven’t looked at the code for a while. The path method is now called query in RTB10.  Does that help?  If you’re willing to renovate it I’d be happy to make it part of the distro again.  I’m assuming you got this from GitHub?

peter

gabestreit, 2019-02-07:
That does help. Thanks. I did get this from GitHub. Unfortunatly I'm relatively new to coding, so this is a bit over my head to renovate.