petercorke/robotics-toolbox-matlab #43: Error in SerialLink/plot3d (line 154)

oladayosolomon, 2019-02-06:
Index exceeds matrix dimensions.

Error in SerialLink/plot3d (line 154)
            opt.ws(5) = mn(3);

Dps8, 2019-02-06:
Does this error only occur in this version of the robotic toolbox?

petercorke, 2019-02-07:
Please send a complete example that demonstrates the error, otherwise all I can say is it works for me.

Dps8, 2019-02-07:
[image]

Dps8, 2019-02-07:
This is the error he gives me. Now I know for your classes, Professor Corke, that does not happen to you. Any suggestions

petercorke, 2019-02-07:
Your query comes via GitHub so I guess you installed by cloning the repo?  The path setup is not quite right.  Did you run startup_rtb first?

peter

Dps8, 2019-02-07:
I did not clone the repos, I used the * .mltb file to install the toolbox in Matlab.

Dps8, 2019-02-07:
Do I have to use the startup_rtb command anyway? Because other commands do work for me without using this one.

petercorke, 2019-02-07:
Interesting.  I’ll have a look at this over the weekend, am travelling this week.

petercorke, 2019-02-07:
If you install from the mltbx file the paths should be setup.  There is also a script called startup_rvc which should do the job for all my toolboxes that are installed.

Dps8, 2019-02-08:
ok professor, I wait for your answer, I'm going to try cloning the repos to see if the error persists. Please you contact me if you discover the cause of the error. Greetings from Cuba.

petercorke, 2019-02-09:
Please remove the MATLAB Phased Array Toolbox from your installation, or else use path tool to put it absolutely last in the path.  That will fix your problem, but also eliminate many other possible errors.

Dps8, 2019-02-10:
thank you very much profesor, it works well.

closed 2019-02-10 by petercorke