petercorke/robotics-toolbox-matlab #44: Display * .stl file correctly using RTB

Dps8, 2019-02-20:
I'm doing the 3D design of a cylindrical robotic arm and I want to import the CAD to Matlab to work with plot3d. What requirements should the * .stl files have to be able to visualize the arm in Matlab?
How do I do it?

petercorke, 2019-02-23:
Their coordinate system should be the link coordinate frame as determined by the Denavit-Hartenberg model of the robot.  They should be named linkN.stl where N is 0, 1, 2, etc.  Let me know how it goes.

Dps8, 2019-02-26:
En que escala trabaja RTB las dimenciones del brazo robótico?The problem is that when I load a link and plot it does not show up well. Is it a problem of scale or is it some DH parameter that I declared wrong?

petercorke, 2019-03-02:
The scale/dimensions of the STL model must be consistent with the lengths in the DH parameters.  ie. if the DH parameters d, a are in metres, then the STL should be in metres.

Dps8, 2019-03-05:
[image]

Dps8, 2019-03-05:
If I declared the limits of the prismatic articulation, why do I have to declare the limits of the workspace?

petercorke, 2019-03-05:
that’s a bug

Dps8, 2019-03-06:
[image]

Dps8, 2019-03-06:
Professor, do you know what could be the cause of this error?