petercorke/robotics-toolbox-matlab #61: ctraj function

NataCalc, 2019-08-19:
Hi Peter,

I am using ctraj function to interpolate my data (Transformation matrix of displacement and rotation). Is there the way to improve the interpolation from step 1 to step 5 performed by ctraj function? 

[image]
[image]
[image]
[image]
[image]

Cheers,
Nataliya

petercorke, 2019-09-29:
sorry for delay, been on holiday... It's not clear what you're trying to improve, can you give me more details.  The pictures don't really help me, sorry.

NataCalc, 2019-09-30:
> Reply to: petercorke/robotics-toolbox-matlab <reply@reply.github.com>

Dear Peter,

I am using ctraj to interpolate my data which was derived from 4DCT scan of two bones motion in the wrist. When rotation of bones becomes significant, ctraj function interpolates in the way where bones start moving unphysiologically. I am wondering to know if there is the way to avoid this through using some control parameters?

Regards,

Nataliya

petercorke, 2019-09-30:
So the initial and final orientation are correct, but the intermediate rotations are not what you want?

NataCalc, 2019-09-30:
Yes, intermediate rotation and displacement are not correct.

Date: Monday, 30 September 2019 at 4:26 pm

petercorke, 2019-09-30:
ctraj does the following:
- linear path between start and end point
- quaternion slerp interpolation of orientation between start and finish

There are an infinite number of paths between two poses and this one is sort of minimum motion.  I think the only way to get the motion you need would be to:

- spline between known good via points
- introduce kinematic constraints between the bones, perhaps one rolls over the other…

NataCalc, 2019-10-01:
Hi Peter,

By ‘spline between start and end point’, did you mean spline interpolation (https://en.wikipedia.org/wiki/Spline_interpolation ). If so, do you have already existing code in robotic toolbox matlab?

Regards,
Nataliya

Date: Monday, 30 September 2019 at 4:52 pm

petercorke, 2019-10-02:
The closest I have to spline code is mstraj() but it interpolates vectors not poses, so you’d have to use XYZ+RPY representation or some such.

NataCalc, 2019-10-03:
Hi Peter,

By XYZ+RPY did you mean the vector  = translationin X,Y, Z + Roll Pitch Yaw XYZ?

Also, in traj = mstraj(p, qdmax, tseg, q0, dt, tacc, options), is p the matrix which consists of trajectory vectors associated with beginning and end of poses? Do you have in your online tutorials explanation of mstraj function?

Cheers,
Nataliya

Date: Wednesday, 2 October 2019 at 10:56 am

petercorke, 2019-10-03:
The documentation help mstraj will give you the details.  Then convert the interpolated XYZRPY vectors back to poses.  Also covered in chap 3 of my book.

closed 2019-11-19 by petercorke