petercorke/robotics-toolbox-matlab #62: The Panda model DH parameters is not correct

vpadois, 2019-09-04:
Dear Peter,

First thanks for this great toolbox which I have been using for years.

After playing around with the panda model, the DH parameters as defined in the mdl_panda.m file seem to be wrong. They are actually different from the source you cite (http://www.diag.uniroma1.it/~deluca/rob1_en/WrittenExamsRob1/Robotics1_18.01.11.pdf) and using the parameters provided in table 1 of this document, the model end-up being correct.

Let me know if I can further help.

best regards
Vincent Padois

petercorke, 2019-09-29:
The parameters I've used seem to have come from https://frankaemika.github.io/docs/control_parameters.html but this page is not completely clear about whether it uses standard or modified DH.  The U.Roma page is clear, by the subscripts, that it is standard DH which is what RTB defaults to.

petercorke, 2019-11-03:
The mdl_panda file has been updated and on a few spot checks seems correct.  Would be happy for additional checking.  Also now includes all rigid-body inertial parameters.

vpadois, 2019-11-05:
I went through your file and some of the dynamic parameters do not correspond to what can be found here : https://github.com/marcocognetti/FrankaEmikaPandaDynModel/blob/master/pdf/RA-L_2019_PandaDynIdent_SUPPLEMENTARY_MATERIAL.pdf (see last page).

I think the attached file fixes this.
[mdl_panda.m.zip](https://github.com/petercorke/robotics-toolbox-matlab/files/3807644/mdl_panda.m.zip)

petercorke, 2019-11-16:
Specifically, which dynamic parameters do you think  are wrong?  One of the authors sent me the LaTeX source of that supplementary material and I scripted the parameter extraction from that.  It seems to me that you have the 6 inertial parameters in the wrong order for RTB, the manual says:
  - The link inertia matrix (3x3) is symmetric and can be specified by giving
    a 3x3 matrix, the diagonal elements [Ixx Iyy Izz], or the moments and products
    of inertia [Ixx Iyy Izz Ixy Iyz Ixz].

The 3 diagonal elements come first, then the products.

vpadois, 2019-11-25:
I actually followed the convention of the RBT when I first created the model but I did not understand correctly the information from one of the authors of the paper. At that time the supplementary material paper was not online yet and I extracted the parameters from some C++ code that was available on the same github repo (that C++ code is no longer there).

Anyway, when I compared both models weeks later I had forgotten about all this and my interpretation of the mismatch was wrong.

Conclusion: my bad, your version is correct.

closed 2019-11-25 by vpadois