petercorke/robotics-toolbox-matlab #63: Julia package port

aminya, 2019-09-10:
Is there any plan for porting this toolbox to Julia language?

petercorke, 2019-09-23:
A student of mine made a start on this, see

https://github.com/thomper/RoboticsToolbox.jl

Julia is a very interesting language and it would be great to complete this.  But I’m struggling just to keep the MATLAB version ticking along.

aminya, 2019-09-23:
That has these functions:
```
rot2, rotx, roty, rotz
trot2, trotx, troty, trotz
se2, se3
r2t, t2r
rpy2r, rpy2t, rpy2jac, tr2rpy
```
To revive that project, it is needed to update the syntax from Julia 0.6 to Julia 1.2.
 There's already a handful of robotics packages in Julia: http://www.juliarobotics.org/ and https://github.com/JuliaRobotics .

So, I was thinking of moving some useful parts of Robotics toolbox to that collection and move it to Julia. So not everything is needed to be transferred.

petercorke, 2019-09-29:
Feel free to update and integrate into JuliaRobotics.  Let me know when it's done.

closed 2020-05-23 by petercorke