petercorke/robotics-toolbox-matlab #66: maybe install error

suk8267, 2019-10-24:
when i tried to install robotics toolbox, so i installed mltbx file and zip file.
and i just followed process in homepage, but it was failed. i check the path, addon manager and so on..
for example, my code is
%%
%%%robotics toolbox

clear all;
close all;
clc;

D2R = pi/180;
R2D = 180/pi;

%link length
length1 = 0.375;
length2 = 0.2175;
length3 = 0.4;
length4 = 0.1975;
length5 = 0.45;
length6 = 0.07;

%initial q
q(1) = 180 * D2R;
q(2) = 90 * D2R;
q(3) = -90 * D2R;
q(4) = 0 * D2R;
q(5) = 0 * D2R;
q(6) = 0 * D2R;

%link(theta d a alpha)
L1 = Link([0 length1 0 pi/2], 'standard');
L2 = Link([0 length2 length3 pi], 'standard');
L3 = Link([0 length4 0 -pi/2], 'standard');
L4 = Link([0 length5 0 pi/2], 'standard');
L5 = Link([0 0 0 -pi/2], 'standard');
L6 = Link([0 length6 0 0], 'standard');

%make serial link
Link1 = SerialLink(L1);
Link2 = SerialLink(L2);
Link3 = SerialLink(L3);
Link4 = SerialLink(L4);
Link5 = SerialLink(L5);
Link6 = SerialLink(L6);

%make robot included virtual link
robot = Link1 * Link2 * Link3 * Link4 * Link5 * Link6;
% figure(1);
% robot.plot(q,'noname', 'jointdiam',2)

%%

%position, forward kin (global frame)
ps = [0; length2-length4; length1+length3+length5+length6];
Ms = [eye(3,3) ps; [0 0 0] 1];

%initial theta, desired theta
thetalist0 = [0;0;0;0;0;0] * D2R;
thetalist = thetalist0;
thetalistsd = [40;41;42;43;45;46] * D2R;

%angular vel & cross product
wz = [0; 0; 1];
wy = [0; 1; 0];
wx = [1; 0; 0];
w = [wz wy wy wz wy wz];
wzhat = [0 -1 0; 1 0 0; 0 0 0];
wyhat = [0 0 1; 0 0 0; -1 0 0];
wxhat = [0 0 0; 0 0 -1; 0 1 0];
what = [wzhat wyhat wyhat wzhat wyhat wzhat];

%e^(w x theta), rotation matrix
R = Rotm(thetalist);
Rsd = Rotm(thetalistsd);

%position vec (body frame)
p1 = [0; length4-length2; -(length1+length3+length5+length6)];
p2 = [0; length4-length2; -(length3+length5+length6)];
p3 = [0; length4; -(length5+length6)];
p4 = [0; 0; -(length5+length6)];
p5 = [0; 0; -length6];
p6 = [0; 0; 0];
pb = [p1 p2 p3 p4 p5 p6];

%vel (body frame, -w x p)
for i = 1 : size(thetalist)
    k = 3 * i;
    v(:,i) = -what(:, k-2:k) * pb(:,i);
end

%body twist
btwist = [w; v];

%%

%forward kin of initial & desired position (body frame) - using function
T = FKinBody(Ms, btwist, thetalist0);
Tsd = FKinBody(Ms, btwist, thetalistsd)

%calculation of forward kin using body twist
TT = FKinB(thetalist, thetalist0, what, v, w, Ms);
TTsd = FKinB(thetalist, thetalistsd, what, v, w, Ms)

%%
%error boundary

eomg = 0.01;
ev = 0.001;

i = 0;
err = 1;
num = 0;

% body twist (6x1 mat)
Vb = BodyTwist(T, Tsd);

while err
    
    num = num + 1;
   
%     thetalist = thetalist + InvJacobBody(thetalist, thetalistsd, what, w, v, Ms) * Vb;
    thetalist = thetalist + pinv(JacobianBody(btwist, thetalist)) * Vb;
       
    TT = FKinBody(Ms, btwist, thetalist)
%     TT = FKinB(thetalist, thetalistsd, what, v, w, Ms);
    Vb = BodyTwist(TT, TTsd);
    
    err = norm(Vb(1:3)) > eomg || norm(Vb(4:6)) > ev; 
    
    robot.plot(q + thetalist','noname', 'jointdiam',2);
    
    
    num
    thetalist*R2D 
    
end

thetalistsd*R2D

%%

and my command window said

Index exceeds the number of array elements (0).

Error in tb_optparse (line 268)
                if isempty(val{1})

Error in SerialLink (line 305)
            [opt,arg] = tb_optparse(opt, varargin);

Error in untitled1 (line 37)
Link1 = SerialLink(L1);
 

and i tried to follow this tutorial in https://kr.mathworks.com/help/robotics/ug/build-a-robot-step-by-step.html
but it can't recognize 'rigidBody' function
i don't know why this happened.
my OS is ubuntu 16.04LTS

petercorke, 2019-10-26:
> 
> when i tried to install robotics toolbox, so i installed mltbx file and zip file.
Which version of toolbox did you install?  What version of MATLAB are you running?
> and i just followed process in homepage, but it was failed.

how did the installation fail, what error messages?

please run the following in MATLAB and reply with the results:

>> rvccheck
>> which tb_optparse
> i check the path, addon manager and so on..
> for example, my code is
> %%
> %%%robotics toolbox
> 
> clear all;
> close all;
> clc;
> 
> D2R = pi/180;
> R2D = 180/pi;
> 
> %link length
> length1 = 0.375;
> length2 = 0.2175;
> length3 = 0.4;
> length4 = 0.1975;
> length5 = 0.45;
> length6 = 0.07;
> 
> %initial q
> q(1) = 180 * D2R;
> q(2) = 90 * D2R;
> q(3) = -90 * D2R;
> q(4) = 0 * D2R;
> q(5) = 0 * D2R;
> q(6) = 0 * D2R;
> 
> %link(theta d a alpha)
> L1 = Link([0 length1 0 pi/2], 'standard');
> L2 = Link([0 length2 length3 pi], 'standard');
> L3 = Link([0 length4 0 -pi/2], 'standard');
> L4 = Link([0 length5 0 pi/2], 'standard');
> L5 = Link([0 0 0 -pi/2], 'standard');
> L6 = Link([0 length6 0 0], 'standard');
> 
> %make serial link
> Link1 = SerialLink(L1);
> Link2 = SerialLink(L2);
> Link3 = SerialLink(L3);
> Link4 = SerialLink(L4);
> Link5 = SerialLink(L5);
> Link6 = SerialLink(L6);
> 
> %make robot included virtual link
> robot = Link1 * Link2 * Link3 * Link4 * Link5 * Link6;
> % figure(1);
> % robot.plot(q,'noname', 'jointdiam',2)
> 
> %%
> 
> %position, forward kin (global frame)
> ps = [0; length2-length4; length1+length3+length5+length6];
> Ms = [eye(3,3) ps; [0 0 0] 1];
> 
> %initial theta, desired theta
> thetalist0 = [0;0;0;0;0;0] * D2R;
> thetalist = thetalist0;
> thetalistsd = [40;41;42;43;45;46] * D2R;
> 
> %angular vel & cross product
> wz = [0; 0; 1];
> wy = [0; 1; 0];
> wx = [1; 0; 0];
> w = [wz wy wy wz wy wz];
> wzhat = [0 -1 0; 1 0 0; 0 0 0];
> wyhat = [0 0 1; 0 0 0; -1 0 0];
> wxhat = [0 0 0; 0 0 -1; 0 1 0];
> what = [wzhat wyhat wyhat wzhat wyhat wzhat];
you use a lot of functions below like Room, FKinBody.  These are not from my Toolbox, where do they come from?
> %e^(w x theta), rotation matrix
> R = Rotm(thetalist);
> Rsd = Rotm(thetalistsd);
>
> %position vec (body frame)
> p1 = [0; length4-length2; -(length1+length3+length5+length6)];
> p2 = [0; length4-length2; [...]

these instructions are for MathWorks toolbox, not my toolbox.  The Robotics Toolbox for MATLAB is open-source software developed by Peter Corke. It is distinct from the Robotic System Toolbox which is developed and maintained by the MathWorks.

suk8267, 2019-10-28:
my robotics toolbox version is 10.3.1 by zip file and my matlab version is 2019a

and i ran the code and result

>> rvccheck
You are using:
 - MATLAB release 2019a
 - Robotics Toolbox for MATLAB 10.3.1 18-Nov-2018 [local (zip,git) install]
 - Machine Vision Toolbox 3.4.0 2015-01-26 [local (zip,git) install]
>> which tb_optparse
/home/hongseok/MATLAB/toolbox/rvctool/common/tb_optparse.m

you use a lot of functions below like Room, FKinBody.  These are not from my Toolbox, where do they come from?

this is only just my fault this is not for robotics toolbox. but when i delete it and run the code, it also give me the same errors

petercorke, 2019-10-28:
Try to remove the vision toolbox and reinstall the robotics toolbox

suk8267, 2019-10-29:
i tried to reinstall the robotics toolbox zip file but website gave me this error '
403 - Forbidden Error
You are not allowed to access this address.
If the error persists, please contact the website webmaster.
'
i could get only mltbx file. so i open the mltbx in matlab . but it is nothing happened

suk8267, 2019-10-29:
woops sorry i just reinstall by mltbx file . it worked! i can use robotics tool box!
thank you very much!!!

closed 2019-10-30 by petercorke