petercorke/robotics-toolbox-matlab #68: The GUI tool pull request.

borninfreedom, 2019-11-28:
Hello, I used Matlab robotics toolbox to make this GUI, https://blog.csdn.net/bornfree5511/article/details/103299633.  It is convenient for users who are accustomed to using GUI. Project address: https://github.com/borninfreedom/robotics_arm/tree/master/RobotStudio. Can you merge this GUI into your project ?

petercorke, 2019-12-01:
Hi Eric,

it’s an impressive piece of work and thanks for sharing it with me. A few comments:

- is this modelled on some other tool?  The interface is not very intuitive to me.  What is the difference between red and yellow GO buttons.
- many of the labels are in Chinese which I cannot read
- on the right hand side, I don’t understand what some of these panels are for, apart from the XY plot
- every axis on every plot should have a label
- how do you handle arm configuration (eg. elbow up/down) for actions that require IK?
- the package is implemented using GUIDE which is, or soon will be, deprecated.  Have you looked at app builder which replaces it?

peter

> Hello, I used Matlab robotics toolbox to make this GUI, https://blog.csdn.net/bornfree5511/article/details/103299633 <https://blog.csdn.net/bornfree5511/article/details/103299633>. It is convenient [...]

borninfreedom, 2019-12-03:
Hi, author. 
- the project is totally based on matlab robotics toolbox. 
- I think many of your questions are because you don't understand Chinese. I will look at app builder and try to re-write the code if I have time. If I do that, I will provide English version.
- I will refine the definition of the axes.
- Sorry, I don't handle the arm configuration yet, thanks for your reminder, I will refine it.
- Thank you for your reply.

petercorke, 2019-12-03:
> 
> Hi, author.
> 
> the project is totally based on matlab robotics toolbox.
thanks, but that wasn’t really my point.  The framework in which you created the GUI is obsolete, app designer is the current one.  It has many advantages but also some limitations, see https://www.mathworks.com/products/matlab/app-designer.html

I’m unwilling to add something based on obsolete technology to the toolbox.
> I think many of your questions are because you don't understand Chinese. I will look at app builder and try to re-write the code if I have time. If I do that, I will provide English version.
This raises the bigger issue about language localisation for MATLAB code, I found

https://blogs.mathworks.com/community/2009/05/04/internationalization-of-guis/

borninfreedom, 2019-12-04:
Thank you very much  for your guidance, I learned a lot . 
```diff
is this modelled on some other tool?
```
- At least in China, I haven't seen similar tools related to robotic arms.
- I will continue to improve it. This is my modern control theory course assignment, and it looks like there are a lot of imperfections.