petercorke/robotics-toolbox-matlab #69: sl_ibvs

gaia442993802, 2019-12-16:
Is the function "slcamera ()" of the camera module in sl_ibvs of the simulink module self-defined or comes with the toolbox?

petercorke, 2019-12-16:
It’s included in the machine vision toolbox.

gaia442993802, 2019-12-16:
OK, I'll try it first

gaia442993802, 2019-12-16:
I tried to execute "sl_ibvs" but still failed. My Matlab version is 2018B update5, RTB version 10.3.1, MVTB version 4.2.1 (shown as 4.3.0 with the "ver" command).The error message are "Error in port widths or dimensions. Input port 1 of 'sl_ibvs/Sum' is a one dimensional vector with 1 elements." and "Error in port widths or dimensions. Output port 1 of 'sl_ibvs/camera' is a [28x1] matrix.".

petercorke, 2020-01-26:
This has been covered a few times in the forum, check out https://groups.google.com/d/msg/robotics-tool-box/EAzY-ciPqoI/uZCjjoWMCQAJ

gaia442993802, 2020-02-12:
Thank you.I will try it.