petercorke/robotics-toolbox-matlab #70: Panda model: qr is not consistent with the joint limits for joint 4

vpadois, 2020-01-30:
Hello,

In the Panda model, joint 4 limits are given by  `'qlim', [-3.0718 -0.0698],` and qr for this same joint is pi/2. Shouldn't qr be specified so that joint limits are actually respected ?

Best
Vincent

petercorke, 2020-02-08:
joint limits are only respected by ikcon().  it would be great to have validation/feedback of this model.  Do you believe the FK is correct?

vpadois, 2020-02-11:
> joint limits are only respected by ikcon().

OK
 
>it would be great to have validation/feedback of this model. Do you believe the FK is correct?

I have used the model quite a bit but I did not perform a systematic comparison with the real Panda model. One of my colleagues and I may soon perform some experiments with the real robot. We can take a bit of time to compare both models in a limited number of configurations. I will keep you posted once done.

closed 2020-02-11 by vpadois