petercorke/robotics-toolbox-matlab #73: plot3d for modified DH

XDinEuro, 2020-03-31:
Is there a special reason for that the plot3d only support standard DH parameters? I was using the plot3d for animate the robot. However, all the meshes are based on modified DH. It would be convenient to plot directly in modified DH parameters so that I don't need to modify the RTB scripts and mesh frame.

petercorke, 2020-05-23:
No reason apart from time.  This is a hobby, so it goes where my interest goes.  If you want to do MDH and send me a pull request I'd be happy to accept it.

QNjeason, 2021-01-02:
I have implemented plot3d based on modified DH.
[image]

QNjeason, 2021-01-02:
https://zhuanlan.zhihu.com/p/340889646