petercorke/robotics-toolbox-matlab #74: Function or variable 'AE' is not defined.

QNjeason, 2020-04-09:
I run the following code with the following error.
`mdl_puma560`
`p = [0.8 0 0];`
`T = transl(p) * troty(pi/2);`
`qr(1) = -pi/2;`
`qqr = p560.ikine6s(T, 'ru');`
`qrt = jtraj(qr, qqr, 50);`
`plot_sphere(p, 0.05, 'y');`
`p560.plot3d(qrt, 'view', ae, 'movie', 'move2ball.gif'); `

Function or variable 'AE' is not defined.
未定义函数或变量 'ae'。
出错 test (line 15)
p560.plot3d(qrt, 'view', ae, 'movie', 'move2ball.gif');

petercorke, 2020-05-23:
where did this code snippet come from?  Make the last line

p560.plot3d(qrt, 'movie', 'move2ball.gif');

closed 2020-05-23 by petercorke

QNjeason, 2020-05-26:
thank you answer.

petercorke, 2020-05-27:
Where did you paste that example from?

QNjeason, 2020-05-27:
I copied it from this page.
https://github.com/petercorke/robotics-toolbox-matlab
[image]
Please also correct the sample program for this problem.

GuarneerFPS, 2022-01-26:
This issue was closed but the wiki remains unchanged. You can find this code snippet under "readme.md".  The first sample code under the title "Code Example" does not work as-is.