petercorke/robotics-toolbox-matlab #78: Bug in the command rpy2r() in version 10.1

Kalana304, 2020-05-21:
Based on the given documentation, **rpy2r()** command computes the angles as radians. 
[image]

I have run the command as following and obtained the following result as shown in the screenshot. 
[image]

Using the basic rotation matric transformations i obtained the following results for the same angle transformation.
[image]

which you can see that the two results are not the same. 

I then ran the **t2rpy()**  to verify what would be the resulting euler angles for the given rotation matrix.
[image]

which means when the **'deg'** option is given the angles are converted to radians but still are considered as degrees. 
Without **'deg'** option in rpy2r(), the output seems to be calculated with degrees which given same answer as to the one obtained from basic rotation matrices. 
[image]

My MatLab version is 2016a. Better to revise the documentation provided and correct it.

Kalana304, 2020-05-21:
I think the followinf script shows the issue.
[image]
the basic rotation matrices take angles as degrees by default. Here the given angles are converted into radians if the opt.deg is True.

petercorke, 2020-05-22:
Check your path, you probably have rotx() coming from the phased array toolbox. This I on the FAQ at the google group support forum.

closed 2020-05-23 by petercorke