petercorke/robotics-toolbox-matlab #80: Franka Panda Dynamic Model

flamelx, 2020-06-01:
Hello, Professor Peter. First of all, thank you for the convenient and easy-to-use robot toolbox.

The following is my problem.I use the model generated by mdl_panda to calculate M, C, G. It should be the same as the result calculated by the code such as get_massMatrix provided by https://github.com/marcocognetti/FrankaEmikaPandaDynModel

But in fact the two matrix values ​​obtained are not the same.

Where is the problem?

petercorke, 2020-06-13:
They should agree, so long as they are all referenced to the same coordinate frames.  Some code gives results referenced to link frames, others to CoG frame.  Can you confirm the frames are the same, and which of the M, C, G matrices are different.

jimi1970, 2022-01-04:
Dear Proffesor Peter, I would like to add my name to this query, or rather I have the following question about the Franka Emika Panda dynamic model. In case I perform the solution of the inverse dynamics problem using Robotics Toolbox  I get different results compared to the solution using the dynamic model published in the following article:
C. Gaz, M. Cognetti, A. Oliva, P. Robuffo Giordano, A. De Luca, 'Dynamic Identification of the Franka Emika Panda Robot With Retrieval of Feasible Parameters Using Penalty-Based Optimization'. IEEE RA-L, 2019. The dynamic model is available here: https://github.com/marcocognetti/FrankaEmikaPandaDynModel/tree/master/matlab/dyn_model_panda  
The problem seems to be in the inertia matrix, or rather in the inertia parameters. It is not clearly mentioned in the paper whether the inertia parameters are relative to the CoM or to the origin of the coordinate systems.  

I am attaching a modified Matlab script demonstrating the different behavior.
Thank you in advance for the answer.

[https://github.com/jimi1970/panda/tree/main/inverse_dynamic_problem](url)

[image]