petercorke/robotics-toolbox-matlab #93: Develpoment for robotics-toolbox-matlab

Sraobot, 2020-11-21:
@petercorke Is there any development taking for the toolbox. Is all the development happening in the python version of the toolbox by Jesse Haviland? I wanted to contribute to the toolbox to the Matlab version of the toolbox as I am more of a MATLAB person than a python programmer. Would you suggest any way to get started to start contributing to the toolbox?

petercorke, 2020-11-22:
Jesse and I are working hard on the Python version, and this consumes all of my cycles which would ordinarily go to the MATLAB version.  At some point in the next 1-2 years I will likely stop supporting the MATLAB version.

You are more than welcome to help out with the MATLAB version, do you have particular things you want to add or fix?

Peter

Sraobot, 2020-11-28:
@petercorke All my enthusiasm for contribution is pretty much finished as you said you will stop supporting it in the future after maintaining it for nearly 3 decades. I think I should make the move to python as you will stop supporting it in the future as mentioned.