petercorke/robotics-toolbox-matlab #97: lspb (Linear Segment with Parabolic Blend) problem

A6GH, 2020-12-26:
s = lspb(s0, sf, m) function call for bounds s0 = 0, sf = 1 and m=36 steps returns the last element of s to be greater than 1.

petercorke, 2020-12-28:
Odd, it's fine for me, using 20b I get:

```
>> tg = lspb(0, 1, 36)

tg =

         0
    0.0018
    0.0073
    0.0165
    0.0294
    0.0459
    0.0661
    0.0900
    0.1176
    0.1488
    0.1837
    0.2222
    0.2643
    0.3071
    0.3500
    0.3929
    0.4357
    0.4786
    0.5214
    0.5643
    0.6071
    0.6500
    0.6929
    0.7357
    0.7778
    0.8163
    0.8512
    0.8824
    0.9100
    0.9339
    0.9541
    0.9706
    0.9835
    0.9927
    0.9982
    1.0000
```

closed 2020-12-28 by petercorke

A6GH, 2020-12-28:
> Odd, it's fine for me, using 20b I get:
>
> ```
>
> tg =
>
>          0
>     0.0018
>     0.0073
>     0.0165
>     0.0294
>     0.0459
>     0.0661
>     0.0900
>     0.1176
>     0.1488
>     0.1837
>     0.2222
>     0.2643 [...]

It probably has something to do with floating point precision. The last number generated is not exactly 1. Try tg(36) == 1.  lspb works fine by itself, but if you call ctraj (which uses lspb) on 36 points, it will report an error when calling trinterp.

petercorke, 2020-12-28:
Definitely a FP issue, it's over by 4 eps which is small in the scheme of things. It's evaluating a quadratic.  Quick and dirty fix would be to clip the value.  I'll add this to the be fixed list, but it won't happen any time soon.  Thanks.