petercorke/robotics-toolbox-matlab #106: Ctraj

lyhbuaa, 2021-11-05:
TC = CTRAJ(T0, T1, N) is a Cartesian trajectory (4x4xN) from pose T0 to T1，now I want to imply a continuous Cartesian trajectory that includes T0,T1,...Tn，Is there a function  to fulfill my needs？ The function  mstraj()  is useful for Joint trajectory but not useful for mine

thank you very much

petercorke, 2021-11-05:
You can use `mstraj` if you encode the SE(3) as a vector, say position and RPY angles

lyhbuaa, 2021-11-06:
Thanks for your reply. But the function mstraj() is not is not an ideal way to interpolate rotation.  In your book Quaternion interpolation is recommend for the rotation interpolatation. So I think  the mstraj() and the Quaternion interpolation should be use together. But I do not know how to do this